0xD34D/ender3-v3-se-klipper-config #7: regarding PR touch

zekromisblack, 2023-11-13:
hey i saw your klipper firmware and config, and i decided to give it a go, how would i level with pr touch?

0xD34D, 2023-11-13:
I added `PRTOUCH_PROBE_ZOFFSET` as the first thing in my `PRINT_START` macro.  You can also add it to your start gcode.

zekromisblack, 2023-11-13:
can i see the code you used for the macro, im having trouble making this work

0xD34D, 2023-11-13:
Here's my macro.  Did you try running the PRTOUCH_PROBE_ZOFFSET from the console to verify that part works for you?

[gcode_macro PRINT_START]
variable_parameter_EXTRUDER_TEMP: 200
variable_parameter_BED_TEMP: 60
gcode:
    G28 ; home all axes
    PRTOUCH_PROBE_ZOFFSET
    M106 S255
    M109 S{params.EXTRUDER_TEMP} ; set extruder temp
    M190 S{params.BED_TEMP} ; set bed temp
    G92 E0 ; Reset Extruder distance to 0
    # G1 E-3 ; Retracts filament to prevent blobs during probing
    # G92 E0 ; Reset Extruder distance to 0
    BED_MESH_CALIBRATE
    BED_MESH_PROFILE SAVE=KAMP
    G4 S5; wait for heaters to recover
    LINE_PURGE

TheDIYGuy999, 2023-11-13:
What is the content of the PRTOUCH_PROBE_ZOFFSET macro itself? I can't find it.

0xD34D, 2023-11-13:
It's a command that is part of changes I've been making to klipper to get an implementation of auto z-offset working.  The command itself is found here https://github.com/0xD34D/klipper_ender3_v3_se/blob/6fa40591230dc7b7507fe1a5a63354152baea0f7/klippy/extras/prtouch.py#L800

TheDIYGuy999, 2023-11-13:
Thanks for your quick answer. So it is only working in combination with your fork of Klipper.

pwlgrzs, 2023-11-13:
Hey @0xD34D I wanted to give it a try but after compiling your fork I am getting 
```
// Unknown command:"PRTOUCH_PROBE_ZOFFSET"
```
Compilation log:
```
pi@klipper:~$ git clone https://github.com/0xD34D/klipper_ender3_v3_se.git
Cloning into 'klipper_ender3_v3_se'...
remote: Enumerating objects: 33145, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 33145 (delta 30), reused 41 (delta 27), pack-reused 33092
Receiving objects: 100% (33145/33145), 25.33 MiB | 2.79 MiB/s, done.
Resolving deltas: 100% (25021/25021), done.
Updating files: 100% (2124/2124), done.
pi@klipper:~$ cd klipper_ender3_v3_se/
pi@klipper:~/klipper_ender3_v3_se$ make menuconfig
Using default symbol values (no '/home/pi/klipper_ender3_v3_se/.config')
Configuration saved to '/home/pi/klipper_ender3_v3_se/.config'
  Creating symbolic link out/board
Loaded configuration '/home/pi/klipper_ender3_v3_se/.config'
Configuration saved to '/home/pi/klipper_ender3_v3_se/.config'
pi@klipper:~/klipper_ender3_v3_se$ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/dirzctl.o
  Compiling out/src/hx711s.o
src/hx711s.c: In function 'command_config_hx711s':
src/hx711s.c:54:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
   54 |     if (h->hx711_count > 4)
      |     ^~
src/hx711s.c:56:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the '                                                                                                                           if'
   56 |         h->flags = 0;
      |         ^
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f1/system_stm32f1xx.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/stm32f1.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Version: 6fa40591
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
lto-wrapper: note: see the '-flto' option documentation for more information
  Creating hex file out/klipper.bin
```

0xD34D, 2023-11-13:
@pwlgrzs are you flashing your printer with the generated `klipper.bin`?  Also, make sure that your klipper service running on the pi is running from the clone of my klipper repo.  Both have to be in sync with these changes in order for it to work

pwlgrzs, 2023-11-13:
Oh, okay, I am running klipper installed via KIAUH. Will clone your klipper repo in a while and retry.

Regarding following:
```
src/hx711s.c: In function 'command_config_hx711s':
src/hx711s.c:54:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
   54 |     if (h->hx711_count > 4)
      |     ^~
src/hx711s.c:56:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the '                                                                                                                           if'
   56 |         h->flags = 0;
      |         ^
```
I am no programmer but I gave it a try with { } as follows:
```
    if (h->hx711_count > 4) {
        shutdown("Max of 4 hx711");
	h->flags = 0;
	h->rest_ticks = HX711S_SAMPLE_REST_TICKS;
	h->hx711s_timer.func = hx711s_sample_event;
    }
```
and on make retry I got this:
```
pi@klipper:~/klipper_ender3_v3_se$ make
  Compiling out/src/hx711s.o
  Building out/compile_time_request.o
Version: 6fa40591-dirty-20231113_223744-klipper
  Linking out/klipper.elf
lto-wrapper: warning: using serial compilation of 5 LTRANS jobs
lto-wrapper: note: see the '-flto' option documentation for more information
  Creating hex file out/klipper.bin
```

I have no idea what I am doing here ;)

pwlgrzs, 2023-11-13:
Okay, how do I install your forked klipper service? I tried to run scripts/install-debian.sh but it keeps failing when creating virtual env :(

```
###### Updating python virtual environment...
RuntimeError: failed to find interpreter for Builtin discover of python_spec='python2'
```

0xD34D, 2023-11-13:
@pwlgrzs I haven't looked at any of the C source code in depth as everything there seems to be working.  Looking at that code I think the intention is only for the `shutdown` function to be called inside the if statement and everything else should still run after the if.

For using my klipper repo in place of your existing one, I think the easiest approach would be to rename the directories.  I'd `mv ~/klipper ~/klipper_original` and `mv ~/klipper_ender3_v3_se ~/klipper`.  From there you might be able to just restart the klipper service with `sudo systemctl restart klipper` or worse case reboot the pi.

pwlgrzs, 2023-11-13:
Seems KIAUH allows for custom klipper repo so I have removed my installation, changed repo to yours and installed. Service is running, code is yours as prtouch.py in extras folder contains your recent commit, but I am still getting unknown command error.

pwlgrzs, 2023-11-13:
So I recalled that file being there minutes before you posted your comment, I have added include directive and...

```
$ set_offset
// probe at 55.000,44.500 is z=2.525000
// bltouch: z_offset: -2.647
// The SAVE_CONFIG command will update the printer config file
// with the above and restart the printer.
```

Very well done!

pwlgrzs, 2023-11-13:
Oh well, almost:
```
Option 'z_offset' in section 'bltouch' must have minimum of 0.0

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

Seems like saving ain't a good idea

0xD34D, 2023-11-13:
\o/ Sweeeet!  I'm happy to hear it worked as advertised for you.  I've never tried saving the config, but I think I can address that in a future patch.  Thanks for the feedback.

pwlgrzs, 2023-11-13:
How to approach this now, change z_offset to 0 and let the macro run on each print?

pwlgrzs, 2023-11-13:
I found this reddit thread https://www.reddit.com/r/klippers/comments/s6pqoe/no_matter_what_i_do_it_just_will_not_let_z_go/
with following comment:
```
To be clear, a Marlin Z Offset of, say -15, would be a Z Offset 15 in Klipper. In other words, Klipper's Z Offset is how much lower the probe tip is (when deployed) than the nozzle tip.
```
Now the good question if I should try printing with macro :D

0xD34D, 2023-11-13:
You can print with that and it should work out.  I found out while testing that I have to negate the value in the probing code, otherwise it was printing that much higher from the bed.  At least it wasn't the other direction ;).

0xD34D, 2023-11-13:
@pwlgrzs My approach when using this is to set the z_offset to 0 in my config

0xD34D, 2023-11-13:
I do need to test the case where the user has set it to non-zero as well

pwlgrzs, 2023-11-13:
Great thanks, I have sent a benchy out.

One thing I noticed during first macro run was that there was a noticeable press from the nozzle on the bed when probing. I would need to flash stock to compare if that was the case there as well.

pwlgrzs, 2023-11-13:
> I do need to test the case where the user has set it to non-zero as well

I will test tomorrow.

pwlgrzs, 2023-11-13:
Benchy failed :(

```
 23:48:46  File opened:CE3V3SE_3DBenchy.gcode Size:3589323
23:48:46  File selected
23:58:03  // probe at 55.000,44.500 is z=2.545000
23:58:16  // bltouch: z_offset: -2.639
// The SAVE_CONFIG command will update the printer config file
// with the above and restart the printer.
23:58:16  !! Error on 'M109 S': unable to parse
23:58:16  !! Error on 'M109 S': unable to parse
23:58:17  echo: Extruder not hot enough
```

```
M109 S{params.EXTRUDER_TEMP} ; set extruder temp
```

I copied macro from above comment:
```
[gcode_macro PRINT_START]
variable_parameter_EXTRUDER_TEMP: 200
variable_parameter_BED_TEMP: 60
gcode:
    G28 ; home all axes
    PRTOUCH_PROBE_ZOFFSET
    M106 S255
    M109 S{params.EXTRUDER_TEMP} ; set extruder temp
    M190 S{params.BED_TEMP} ; set bed temp
    G92 E0 ; Reset Extruder distance to 0
    # G1 E-3 ; Retracts filament to prevent blobs during probing
    # G92 E0 ; Reset Extruder distance to 0
    BED_MESH_CALIBRATE
    BED_MESH_PROFILE SAVE=KAMP
    G4 S5; wait for heaters to recover
    LINE_PURGE
```

I use CuraSlicer if that matters.

0xD34D, 2023-11-13:
Yeah, that was just an example of where I placed the command but that's what I call from my slicers start g-code.  You should add `PRTOUCH_PROBE_ZOFFSET` to your start g-code in cura.

zekromisblack, 2023-11-14:
ive been trying to manually use ``PRTOUCH_PROBE_ZOFFSET`` through the console but ive been getting errors such as the one below
[image]

pwlgrzs, 2023-11-14:
> Yeah, that was just an example of where I placed the command but that's what I call from my slicers start g-code. You should add `PRTOUCH_PROBE_ZOFFSET` to your start g-code in cura.

So, after few prints:
- it works very well in general, no issues so far
- seems calculated offset is bit too low, on both PLA and PETG bottom benchy text was not or barely visible (it was worse on PETG)

Do you know if there is a way to compensate in slicer? I am testing Cura and Prusa so if you know for any of these that would be great.

hafeez-rehman, 2023-11-15:
@0xD34D Great work, I merged your code into my local klipper code, built it, flashed it. Everything just worked perfectly on first try. First layer is perfect. Only issue so far is calculated value should be positive, not a big deal. Thank you so much.

futurewolf70, 2023-11-16:
@pwlgrzs I think if you add "SET_GCODE_OFFSET Z=(the distance between the calculated offset and the correct offset)" in you start GCODE after the macro, you should be able to correct the offset. As long as it's consistently off by the same amount.

Be ready to shut it down if the offset is wrong in order to protect your bed. Of, if you're like me, you have a bunch of previously wrecked beds lying around that you can test it on.

pwlgrzs, 2023-11-19:
I do not know if this is useful information in anyway but who knows.

SonicPad z-offset value:
```
#*# [bltouch]
#*# z_offset = 2.579
```

Your klipper z-offset value:
```
#*# [bltouch]
#*# z_offset = 2.621
```

I guess that's just FYI

mshrem, 2023-11-20:
> [gcode_macro PRINT_START]
> variable_parameter_EXTRUDER_TEMP: 200
> variable_parameter_BED_TEMP: 60
> gcode:
>     G28 ; home all axes
>     PRTOUCH_PROBE_ZOFFSET
>     M106 S255
>     M109 [...]

Hello sorry for the question , i am a newbie with a first 3d printer , i have managed to copy your repo and flash the firmware onto the printer , i am able to run the  PRTOUCH_PROBE_ZOFFSET and returns a g28 command. 

I have also added the print start macro as you put in here , but you are suggesting we add this information into cura ? 

I dont want to send a print test yet, want to confirm i need to add this information into Start G-code in cura ?? How about End G-Code ???? 

[image]

Can anyone provide me with some examples for cura , will be greatly appreciated. 

Thanks

0xD34D, 2023-11-22:
@mshrem after looking at your start GCODE from Cura, I would add `PRTOUCH_PROBE_ZOFFSET` right after the G28 homing command.

shebang79, 2023-12-09:
Thought I'd give this a shot, made the klipper.bin just fine, but I cannot get it to flash for the life of me. I've tried multiple sdcards (4096,etc) and it just will not flash no matter how many times I try. Pulling my hair out here. 

Can I flash this from kiauh? Only issue with this, is it asks me which MCU it wants and I'm not sure which is what. ttyS0 or ttyAMA0?

shebang79, 2023-12-09:
> That command is defined in `prtouch.py` https://github.com/0xD34D/klipper_ender3_v3_se/blob/master/klippy/extras/prtouch.py#L800. Did you add the `prtouch.cfg` configuration file from this repo and `[include prtouch.cfg]` in your printer config? I'm still figuring out how a lot of the internals of klipper work but I think it won't load the prtouch module unless it's in the printer configuration.

prtouch.cfg? I can't find it anywhere? It doesn't seem to be in this repo....?

shebang79, 2023-12-09:
> check again, there are only 4 files in the repo and it's one of them...

Sorry I don't understand what you mean, only 4 files?

[image]

csutcliff, 2023-12-09:
That's not this repo

shebang79, 2023-12-09:
> That's not this repo

lol yes I figured that out and edited my comment -- quite right. It's all good now, thanks.