0xD34D/ender3-v3-se-klipper-config #10: mcu

Alexandrinn, 2023-12-29:
please add in the comment to the printer configuration that when creating the firmware in Klipper Firmware Configuration you need to activate [*] Disable SWD at startup option

bootuz-dinamon, 2024-01-11:
Wrong USART.
You need USART1 PA10/PA9 for Ender 3 V3 SE type-c connect.

Alexandrinn, 2024-01-12:
> Wrong USART. You need USART1 PA10/PA9 for Ender 3 V3 SE type-c connect.

The images are not mine, I took them only for a visual example... The author has already added this item to the file, so I think the question is closed

alex-elivate, 2024-05-20:
I didn't have to disable SWD, but I did need to flash my firmware twice.  The first one looked and acted like it took, but wouldn't connect.  I did a make clean and with the same config did another make, re-flashed it, and it worked right away. I'm wondering if this may cause some false starts for some people.
Anyway, success here.  I just had to adjust my stepper_x positions to -10 so it wouldn't print off the bed to the left. Great work on this.  I'm thrilled.