0xD34D/ender3-v3-se-klipper-config #15: Max Speed

blackadar, 2024-01-23:
I noticed you set the maximum speed and acceleration to:

```
[printer]
kinematics: cartesian
max_velocity: 150
max_accel: 3000
max_accel_to_decel: 3000
max_z_velocity: 50
square_corner_velocity: 5.0
max_z_accel: 500
```

Creality says the printer can print up to 250 mm/s and accelerate up to 2500 mm/s^2. Would the better config not be:
```
[printer]
kinematics: cartesian
max_velocity: 250
max_accel: 2500
max_accel_to_decel: 2500
max_z_velocity: 50
square_corner_velocity: 5.0
max_z_accel: 500
```

Just wondering how you decided on these numbers. Thanks for sharing your work with the community! :)

0xD34D, 2024-01-24:
Hey @blackadar,

It's been a while so I can't recall my rationale at the time, but these are very conservative values for this printer.  I personally use a speed of 500 mm/s and an acceleration of 10000 mm/s² on mine.

I need to review creality's configs that they have for the sonic and nebula pads and I'll probably update those config with their values.

Thanks for bringing this to my attention.

0xD34D, 2024-01-24:
Actually I take my last comment back.  I just checked and my values are correct https://github.com/0xD34D/ender3-v3-se-klipper-config/blob/e9c5a9d043183f37588c10c9d033afc38b60c522/printer-creality-ender3-v3-se-2023.cfg#L126

I'm not quite sure where you found the 150 mm/s speed but it's not currently in my config ;)

closed 2024-01-24 by 0xD34D

blackadar, 2024-01-24:
Oh whoops! You're right, looking back at my notes I got them [here](https://github.com/bootuz-dinamon/ender3-v3-se-full-klipper/blob/main/printer.cfg). Sorry for the confusion - I think I'll be switching to your config 😅