0xD34D/ender3-v3-se-klipper-config #17: Filament sensor support?

pellcorp, 2024-03-13:
Will a filament sensor connected to the main board work?   Or do I need to connect it to the rpi?

0xD34D, 2024-03-13:
Filament sensor works connected to the main board.  There's a connector on there that is marked "Filament".
I also have a config for using it https://github.com/0xD34D/ender3-v3-se-klipper-config/blob/main/filament_sensor.cfg

pellcorp, 2024-03-15:
ah awesome thanks so much

closed 2024-03-15 by pellcorp

wrobelda, 2024-04-01:
@0xD34D  Did you use a standard Crealty detector? It's said to have 3 pins, meanwhile the motherboard has extra 24V fourth pin, but it's safe to ignore it?

0xD34D, 2024-04-02:
> @0xD34D  Did you use a standard Crealty detector? It's said to have 3 pins, meanwhile the motherboard has extra 24V fourth pin, but it's safe to ignore it?

I used a generic 3 pin sensor and it was fine to just connect +5V, GND and the signal pin.

Haleysoftware, 2024-04-10:
The Crealty filament sensor that I just bought came with a 3 to 4 pin extension cable. Not sure when they started including it, but it seems like it might have been around the KE release.