0xD34D/ender3-v3-se-klipper-config #30: Stm32F401

Klumba93, 2025-04-25:
Not long ago, a new revision of boards for ender 3 v3 se with a processor appeared Stm32F401
I tried a bunch of options for flashing the MCU but still couldn't find the right way to flash it. I would be grateful if someone could figure this out and give me the parameters for compiling the firmware.

Argonlifesee, 2025-08-24:
Original motherboard:
- Micro-controller architecture: STMicroelectronics STM32
- Processor model: STM32F103
- Bootloader offset: 28KiB
- Communication interface: Serial (on USART1 PA10/PA09)

New motherboard:
- Micro-controller architecture: STMicroelectronics STM32
- Processor model: STM32F401
- Bootloader offset: 64KiB
- Communication interface: Serial (on USART1 PA10/PA9)

This should work. For a complete guide you may want to Look at the [guide from athemis](https://athemis.me/projects/klipper_guide/)

timdodge, 2025-08-31:
Hi, I have a v3 se with the STM32F401processor. I used Athemis' guide for the initial firmware install - all worked great.

However, I'm trying to update to this firmware using the same instructions, but the new firmware doesn't want to flash - I've tried putting the klipper.bin in both the root dir and STM32F4_UPDATE directories, but no joy.

fluidd is saying:
```
MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: 5eabae67
MCU(s) which should be updated:
mcu: Current version v0.13.0-154-g9346ad19
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'mcu': Unknown command: run_dirzctl
```

My new firmware definitely has this command:
```
timmy@dennis:~/klipper/out $ strings klipper.elf | grep run_dirzctl
command_run_dirzctl
command_run_dirzctl.lto_priv.0
```
Any ideas?

timdodge, 2025-08-31:
Figured it out - I wasn't aware that the bin filename had to be different to previous updates...

wjaros, 2025-12-03:
did u manage to make it work? I cannot go past the BLTOUCH, just does not want to cooperate